MiSTle-Dev/MiSTeryNano #1: SD card recognition problem

misterjbam, 2023-10-14:
Although I tried with several different SD cards, I was never able to get a floppy disk to work.
no floppy disk is displayed in the menu and no boot from floppy disk.
I tested with several different SDs all in FAT (formatted under Windows and Linux), I even tested in FAT32.
I tested with the last 3 cores and also with 2 different TOS.
Tested by putting only DISK_A.ST on the SD and putting multiple files.
I can't get it to recognize a floppy disk.
My SD drive works fine with Nestang so that's not the problem.

harbaum, 2023-10-14:
The TOS version doesn't matter.

So you get an empty OSD even with a few .st files on the card?

What do the LEDs on the Tang Nano show? The two closer to the USB connector are floppy a and b activity. But in your case the other 4 are more important. They show the SD card state. What do these show?

Also how big is your SD card?

misterjbam, 2023-10-14:
Yes the OSD menu is empty despite the presence of .st file on the SD.

tested with a 2GB card, an 8GB card and a 16GB card

For the LEDs (1 is closest to the USB)
Led 1: Lit at startup, then on the TOS, flashes very quickly and is very weak)
Led 2: off at startup, then on the TOS, flashes very quickly and is very weak)
Led 3: off
Led 4: off
Led 5: off
Led 6: is constantly lit.

harbaum, 2023-10-14:
I assume that LEDs 3 to 5 are all lit for a short moment on startup? So it seems that your card's low level init worked. But the file system cannot be read. 

Do you have a Linux box? And a card reader? Assuming your sd card shows up as /dev/sdb, please do this:

```
$ sudo file -s /dev/sdb
/dev/sdb: DOS/MBR boot sector; partition 1 : ID=0xc, start-CHS (0x1,0,1), end-CHS (0x35f,63,32), startsector 2048, 31127552 sectors, extended partition table (last)
```

This tells us that there's one partition. So then please do:

```
sudo file -s /dev/sdb1
/dev/sdb1: DOS/MBR boot sector, code offset 0x58+2, OEM-ID "mkfs.fat", sectors/cluster 64, reserved sectors 64, Media descriptor 0xf8, sectors/track 32, heads 64, hidden sectors 2048, sectors 31127552 (volumes > 32 MB), FAT (32 bit), sectors/FAT 3840, serial number 0x2361ebc8, label: "ATARI ST   "
```

And this partition can be mounted and has the files:
```
sudo  mount /dev/sdb1 /mnt/
ls /mnt/
ATB.st disk_a.st
$ sudo umount /mnt 
```

Something must be fundamentally different with your cards.

misterjbam, 2023-10-14:
Thanks for your help .
here are the results I got

`sudo file -s /dev/sdb
/dev/sdb: DOS/MBR boot sector; partition 1 : ID=0x6, start-CHS (0x4,4,1), end-CHS (0x3ff,254,2), startsector 2048, 8386560 sectors`

`sudo file -s /dev/sdb1
/dev/sdb1: DOS/MBR boot sector, code offset 0x3c+2, OEM-ID "mkfs.fat", sectors/cluster 128, reserved sectors 128, root entries 2048, Media descriptor 0xf8, sectors/FAT 256, sectors/track 62, heads 246, hidden sectors 2048, sectors 8386554 (volumes > 32 MB), serial number 0x3b7dee0c, label: "TANG       ", FAT (16 bit)`

misterjbam, 2023-10-14:
yeahhhhhhh
that's good, it works

closed 2023-10-14 by misterjbam

harbaum, 2023-10-14:
> the problem came from the size of the clusters: 64kb

128 sectors per Cluster should have worked. But I never tested that. I'll take a look at that.

Even your 4GB card used that many sectors per Cluster? That's surprising. On smaller cards the clusters are usually also smaller.