MiSTle-Dev/MiSTeryNano #2: GPIO normalization

misterjbam, 2023-10-15:
Good morning

Would it be possible to standardize the GPIO port?
For the joystick port of this Atari ST core, is it possible to move it to use the same location and controller as the core NESTANG, PACMAN, GALAGA, ... ?
And for futur the keyboard, would it be possible to standardize this with that of the core C64?
If all cores is standardized it will be simpler for users.

Thanks to all the developers

harbaum, 2023-10-15:
NESTANG uses those game pads that come with the board. I don't have any of these.

And there's a c64 core? Where?

I actually plan to remove all that and make it optional and instead go for the USB-C already present on the Tang Nano 20k.

misterjbam, 2023-10-15:
You would have to open a prize pool to offer you the same controller as Nestang

c64 : https://github.com/vossstef/tang_nano_20k_c64

nice idea to pass everything through usb

harbaum, 2023-10-15:
This c64 is using PS2 keyboards. We have 2023 ...

misterjbam, 2023-10-15:
I agree with you, using USB is really better

harbaum, 2023-10-17:
Since the original MiSTery core uses PS2 like signals internally it may make sense to temporarily add PS2 to MiSTeryNano as well. This would allow to receive PS2 like mouse and keyboard signals from a USB keyboard and mouse. Or preferably from one of those tiny USB keyboard touchpad combo devices.

harbaum, 2023-10-25:
One day we'll think about this. But currently the setup changes every few days and it doesn't make much sense to try to establish a standard now.

closed 2023-10-25 by harbaum