MiSTle-Dev/MiSTeryNano #3: Stock BL616 firmware?

rog77, 2023-10-26:
I read on the forum that there is talk of you being able to maybe distribute the TN20K/BL616 stock firmware, at least in binary form, and hopefully get permission to distribute source code - has there been any progress on this please?

harbaum, 2023-10-27:
I am not allowed to distribute anything. This is all Sipeeds decision and it would imho be best if they at least publish the binary on their wiki/GitHub pages and even better if they also release the source code. This will make the Tang Nano 20k much more useful and would actually give its "UPDATE" button a reason to exist.

rog77, 2023-10-27:
Thabks for the responce. I don't suppose you would be able to share the contact details of where you asked for it? Or... for preference,  tag them into this thread if they are on github?

I think an open BL616 firmware that allows flashing from a PC, as well as USB Host, and maybe flash from SD card into FPGA SRAM would really open this platform up for retro enthusiasts - sipeed obviously have this realm of thought in mind, given decisions made about the platform. But it won't take off until that firmware is either available or open source.

Thanks for your efforts on this project though, you have shown them the way :-)

harbaum, 2023-10-27:
I'd suggest you just open an issue for the M0S/BL616 at Sipeed. It seems https://github.com/sipeed/M0S_BL616_example might be one repository that's somehow related.

rog77, 2023-10-27:
Thanks for the response, I don't see them being very responsive on that repository, but I remembered that I had an email contact for someone on Sipeed staff, so have emailed them directly. If they respond I'll also have a go.at suggesting the potential commercial benefits of them releasing the source code under an open licence. Cheers!

harbaum, 2023-10-28:
You could also visit the Sipeed or Tang Nano telegram channels.

rog77, 2023-10-28:
Might I ask, presuming you are allowed to discuss the code, is there much similarity to this:
https://github.com/sipeed/RV-Debugger-BL702

Based on the fact that it does usb2uartjtag and is also a BL chip, although I appreciate it doesn't do the PLL or have QSPI. If it was similar enough, could this be combined with your USB Host/hid code to achieve both sets of functionality on the BL616 without breaking any agreement with sipeed?

Personally, I'd love to see micropython on this (BL616) device, but that seems a bit optimistic for now.

harbaum, 2023-10-28:
At a quick glance it doesn't seem to look very similar.

Bouffalo imho has some python for the bl616 in their sdk which may actually be derived from Micropython.

rog77, 2023-10-28:
It's pikapython, I asked the devs and they were kind enough to implement USB Serial Repl for it, as opposed to the default UART pins, but I don't think the python is mature, based on the number of fixes I see going through.

Thanks for the response on the similarity though, I thought was worth asking just in case.

Out of curiosity, do you think the BL616 could run your code, a version or the uploader, and some micropython variant in the same firmware to make it multifunctional? I have raised this idea with sipeed previously,  for reasons related to projects such as Mister variants.

vossstef, 2023-11-05:
hi, there is a project for Gowin JTAG programming using the CH552T µC that had been used on the very first Tang Nano 1k 2704 revision (still with 24Mhz crystal and CH552T µC, https://dl.sipeed.com/shareURL/TANG/Nano) . At that time the µC firmware didn't even supported UART to FPGA which had been added by Kongou Hikari.  I reflashed my old 1K TN and UART is'Ts working like a charm both in programming using Gowin GUI and USB UART interface. Maybe helpful to keep JTAG function in the BL616 apart from the HID device functions. 
https://github.com/diodep/ch55x_jtag 
https://qiita.com/ciniml/items/05ac7fd2515ceed3f88d

harbaum, 2023-11-05:
Since the JTAG signals on the TN20k are exposed on solder pads it should be possible to connect any uC and give a custom flasher firmware a try.

rog77, 2023-11-05:
https://github.com/emard/esp32ecp5/blob/master/circuitpython/jtag.py

I was pondering something like this on the BL616, like run a menu to pick a bitstream from SF and flash it to SRAM without needing any other hardware - only a modification to the stock firmware.

rog77, 2023-11-05:
> hi, there is a project for Gowin JTAG programming using the CH552T µC that had been used on the very first Tang Nano 1k 2704 revision (still with 24Mhz crystal and CH552T µC, https://dl.sipeed.com/shareURL/TANG/Nano) . At that time the µC firmware didn't even supported UART to FPGA which had been added by Kongou Hikari. I reflashed my old 1K TN with that and it's working like a charm both in programming using Gowin GUI and USB UART interface. Maybe helpful to keep JTAG function in the BL616 apart from the HID device functions. 

^^^

https://github.com/vossstef/tang_nano_9k_6502

Like could you run micropython/pikapython on the BL616, then use the USB hid in combination with the HDMI/VT52 terminal you use in this project to present a menu to the end user? If there is a golden image to do this and allow access to the SD card via BL616, as long as the bitstream can be held in MCU RAM, couldn't it be flashed back to FPGA SRAM? Then on reboot it just goes back to the menu selection bitstream?

harbaum, 2023-11-05:
The bitstream is bigger than the ram of the bl616. So downloading from SD card first won't work.

As a USB host the bl616 can read from USB sticks.

rog77, 2023-11-05:
Interesting! Thanks for setting me straight on the ram issue. I presume it is capable of doing HID at the same time as mass storage?

I really do think it would take off as a platform if it were possible to have a mister style menu. I don't think it would be a big barrier (for end users) to require a USB hub hanging off the power injector/USB adapter, it would still be cheap and cheerful.

harbaum, 2023-11-05:
The MISTer menu is basically the MIST menu. And I took that straight from the Minimig. So that is very old and did not age that well.

A few days ago I got the u8g2 library running on the bl616 and the OSD of MiSTeryNano:

https://github.com/olikraus/u8g2/discussions/2288

This imho is a slightly nicer option for a menu.

rog77, 2023-11-05:
Nice work :-)

Alas I am no C programmer,  hence my interest in using uPython or the like to have the BL616 do JTAG without a PC (yet that is its own rabit hole) but on looking I see https://github.com/HerrNamenlos123/JTAG_Interface/blob/master/src/jtag.c would it be a big adaptation to make that work with gowin chips rather than Intel?

harbaum, 2024-01-16:
The stock firmware is now available as a binary: https://github.com/harbaum/MiSTeryNano/tree/main/bl616/friend_20k

No source code, so no way to extend this. But this at least allows to mess with the internal BL616 without risking to brick the board.

harbaum, 2024-01-30:
Sipeed has recently started to encrypt the firmware on the TN20k. This means that it won't boot any replacement firmware. Even worse: It even won't boot the friend_20k firmware distributed here. I essentially bricked one of my boards by updating the firmware. Currently there's no way back and the board is rendered unusable.

harbaum, 2024-01-30:
I am in discussion with them and they try to find a way for us to load our own code into a seperate flash area other than the area used by the pre-installed firmware. Hopefully that'll work.

fanoush, 2024-02-12:
> Sipeed has recently started to encrypt the firmware on the TN20k.  I essentially bricked one of my boards by updating the firmware. Currently there's no way back and the board is rendered unusable.

Does this affect new boards or also old boards we already have? By 'bricked one of my boards by updating' means you tried some newer version of the firmware with this feature ? Or they changed updating procedure even when reflashing older version too?

I wonder what is their rationale behind this. It is a hobbyist board.

I got my board in December, how do I find out if the firmware is is already encrypted?

fanoush, 2024-02-12:
Thanks. Still I don't understand how you bricked one of your boards. You bought new one and that already came with encrypted firmware? If yes, when did you buy it?

vossstef, 2024-02-12:
Is it possible to check the 'protected' status of the Chip by reading a certain memory location with the FlashCube Toolset (Utils tab) to figure out what board type i do have ?

harbaum, 2024-02-14:
I've got permission to distribute the encypted firmware as well:

https://github.com/harbaum/MiSTeryNano/tree/main/bl616/friend_20k

As stated there, I don't know if it possible to determine the encryption state beforehand. But this allowed me to unbrick my own TN20K this way, so it should now be safe to simply try it. But I won't quarantee anything and you might still brick your device.

harbaum, 2024-02-14:
> Thanks. Still I don't understand how you bricked one of your boards. You bought new one and that already came with encrypted firmware? If yes, when did you buy it?

Exactly. I bought an new board, flashed my own firmware and found that the board wouldn't run it and neither would it run the original firmware I already had. So i contacted Spieed and learned that the recents boards come with encrypted firmware and would only run an encrypted firmware.

fanoush, 2024-02-15:
> I am still discussing how to solve this issue. One idea is to add the ability to store an additional unencrypted firmware together with the encrypted one.

It is strange that there would be no way back. Typically when you protect the flash from reading there is some mass erase command to erase the flash and start from scratch with some full version of firmware. Maybe that unencrypted one is just partial one and the flashing procedure is not erasing everything? the config file has
```
[cfg]
# 0: no erase, 1:programmed section erase, 2: chip erase
erase = 1
```
so value 2 could work?

harbaum, 2024-02-15:
It depends what the vendor tries to achieve. This feature seems to be advertised as "secure boot" which is exactly how this works: There is no way to boot anything than the official firmware.

In this case the idea may be different: You can release firmware updates but prevent people from building their own hardware and run your encrypted firmware. So you are not protecting the firmware from being copied but hardware clones from running your firmware.

harbaum, 2024-02-15:
> so value 2 could work?

I think I tried that without success. But feel free to give it a try. We should have everything needed to return to a sane state.

harbaum, 2024-02-21:
I have noticed that all my boards the come with the encrypted flasher also _don't_ have C51 populated:

https://github.com/harbaum/MiSTeryNano/blob/main/images/tn20k_c51.jpg

fanhuanji, 2024-04-21:
Hi, I've dumped firmware from my chip using 'BouffaloLabDevCube'. It seems there's a lot of segments are similar with the encrypted firmware that you've published. Could you please dump some of your firmware so we can analyze them?

Dumped firmware: (just rename the file extension from zip to bin. I have to rename it in order to upload it)
[flash.trunc.bin](https://github.com/harbaum/MiSTeryNano/files/15050954/flash.trunc.zip)

harbaum, 2024-06-18:
I'll close this. Neither the bouffalo SDK nor the state of the flasher firmware IMHO allows for a satisfying solution.

Either someone spends the time to move the MCU functionality into the FPGA (which will probably need a bigger FPGA to become a satisfying solution) or we'll have to live with an extra MCU. Being able to run on a bare TN20k would be really nice. But it's IMHO currently not worth the hassle.

closed 2024-06-18 by harbaum