MiSTle-Dev/MiSTeryNano #4: Should be PS/2 KB/Mouse support possible without flashing BL616?

TzOk83, 2023-10-28:
It is a little unclear for me - if I just flash the atarist.fs, will I have a PS/2 KB/Mouse support without flashing anything to BL616?

harbaum, 2023-10-28:
You need one BL616 for USB. Either the on-board one or a separate one like the one on the M0S Dock. It then translates from USB to PS2.

If you want to use real PS2 keyboards and mice, then you might actually be able to do so without any BL616 and connect your keyboard and mouse directly (with 5v-to-3.3v level shifters!!!) to the Tang Nano like the authors of the C64-on-Tang-Nano project do. But I have never tested that and it might not work at all.

In the long term the BL616 will also handle the on-screen-display and mass storage (floppies and HDD via SD card). So a BL616 will be needed, anyway.

TzOk83, 2023-11-26:
As the PS/2 support is gone, I'm closing this issue.

closed 2023-11-26 by TzOk83