MiSTle-Dev/MiSTeryNano #5: Combine USB Keyboard Mouse and Joystick into a single firmware ?

vossstef, 2023-11-04:
Dear Till, do you plan combining your USB Joystick (Pacman) with the USB Keyboard + Mouse (ST) function into a single package?
Would be very useful for my c64 / vc20 projects. thx Br, Stefan

harbaum, 2023-11-04:
Yes, of course. The code is actually alreeady there, the joystick signals are just not exposed. The m0s does not have the five more gpios needed for that.

I currently switch from those two ps/2 channels to one fast spi. That''s faster, more flexible, easier to use and bidirectional.

And this will also be able to control a on-screen-display and will allow the user to select floppy or rom  images. You might have a use for that as well.

vossstef, 2023-11-04:
thanks!
I had a lot of trouble with the onboard BL616 SPI signals blocking my digital Joystick inputs. Looks that the original firmware did not released some pins properly sometimes. After migrating the c64 design to your pinmap all fine now.

vossstef, 2023-11-08:
Do you already have any planned use of M0S module edge pins pads 19, 20, 21, 22, 24 respectively GPIO 29, 30, 0, 1, 3 ?
(those aren't available on the M0S dock and would require some soldering skills to make use of it)
Still considering to prepare for some kind of dual footprint Interface (old digital Joystick and USB type converted by bl616 at the same time). Those minimum extra single pin needed for a digital Joystick input could be offered in that sense. 

btw your bl616 USB to PS/2 Keyborad interface works with a small modification in my c64 core. Problem of course in my project...

harbaum, 2023-11-08:
Glad to hear you're having a use for the USB/PS2 stuff.

I actually don't plan to use more pins of the M0S dock. I have recently switched from those two PS2 ports to one SPI port (ss, sck, miso and mosi). It currently runs at 20Mhz and can be used to send all kinds of data into and out of the FPGA incl. e.g. Joystick and on-screen-display. Much like I did 10 years ago on the original MIST.

closed 2023-11-10 by vossstef