MiSTle-Dev/MiSTeryNano #8: No mouse/kb on current release

TzOk83, 2023-11-26:
I've just uploaded filestream and TOS images (1.04UK and 2.06UK) to Tang Nano 20k via OpenFPGALoader (from Windows 10, which was not as easy as it seems). I've also programmed M0S dock with supplied firmware using Bouffalo Flash Cube. Atari boots to the desktop, but there is no keyboard or mouse response.

weav2k4, 2023-11-26:
Just updated to the new release and mouse and kb working fine for me.  I know it can be picky on the mouse and kb but my Rii X8 mini keyboard/trackpad works perfectly with it.  Maybe try different ones?

TzOk83, 2023-11-26:
Yes, today I tried different mice, and one was half-working; it was making only horizontal cursor movements (Logitech M195), but another one (noname mouse) works fine.

The initial problem was, that I built it in the middle of the night after a whole day of work... and the pinout of the Dock is not very intuitive (angled connectors are evil ;) ), so I've managed to put it backward. The good news is, it survived putting 5V to 3V3 line.

I'm leaving this issue, as it may help someone with the OpenFPGALoader issue with the FDTIBUS driver under Windows 10. Apparently, instead of MSYS2/MINGW64, you can also use Windows Subsystem for Linux (Ubuntu on Windows), but you have to update the kernel manually (at least v5.10.60.1): https://www.catalog.update.microsoft.com/Search.aspx?q=wsl

closed 2023-11-26 by TzOk83