MiSTle-Dev/MiSTeryNano #13: USB Joystick not working

TzOk83, 2024-01-05:
I have recently tried my USB joysticks, and one didn't work at all, while another one was almost working - there were some erratic reactions to the d-pad, but no fire button at all.

Please also notice, that the real ST, the right mouse button acts as a joystick fire button (they are physically wired together).

They were both generic gamepads, one not working was analog, with the digital d-pad (PlayStation knock-off), and the one almost working was a digital d-pad only (namely a PiHut SNES USB Pad: [https://botland.com.pl/kontrolery-do-gier-usb/14940-pihut-snes-retro-kontroler-usb-do-gier-zgodny-z-raspberry-pi-604776613015.html].

harbaum, 2024-01-05:
I have a bunch of USB joysticks but haven't tested many of them, yet. This will sure come once all the basic's are implemented.

TzOk83, 2024-01-06:
Is there any way I could help (without experience in Verilog)?

harbaum, 2024-01-06:
The USB part happens completely on the M0S/BL616. There is no Verilog involved.

harbaum, 2024-02-21:
So I do have a quite similar looking cheap SNES like shaped gamepad which just works fine.

HaraldSimon, 2024-06-22:
I have checked two USB joysticks, both not working (Competition Pro/EasySMX).
The USB descriptor is longer than 128 bytes which is a limit inside a structore of the cherry usb stack (report_desc in usbh_hid struct). After patching the sources with the value of 256 both are listed in the debug log of the BL616.
But they are still not working. The Competition Pro does not send usb urbs, the EasySMX does not transfer axes...

harbaum, 2024-06-24:
Development now takes place at https://github.com/harbaum/FPGA-Companion

But the issues are likely the same. Feel free to submit a PR if you fix things.