MiSTle-Dev/MiSTeryNano #14: Odd beahviour with HID devices (Mouse)

dna2496, 2024-01-12:
Hey Hi, i got it working! Its an awesome work you did! Everything is working good so far. I had some research to do cause i did the installation on Windows but with the right tools it was not that hard. I installed several mouse keyboard kombinations. Most as wireless desktop solutions from Logitech, Microsoft and others and the keyboards all work fine only the mouse got a strange behaviour .. some only moved horizontal and some vertical even just a wired Logitech mouse had this behaviour. the only mouse that worked well on its own without a keyboard is an apple magic mouse with this little trackball as mousewheel. If it helps i can write down the models of the tested gear. I´ve try many kombinations with the cables .. short long ... with breadboard whithout ... it doesnt change the behaviour.

harbaum, 2024-01-12:
I also have a mouse that would misbehave somewhat. I'll definitely have a look at that. The main features are now mainly done with harddidk support being nearly finished. Work will thus shift to ironing out such issues ...

dna2496, 2024-01-12:
perfect. no need for hurry :) just wanted to give some infos and if you want to having me to test something or help with something let me know.

dna2496, 2024-01-24:
Perfect i´ll try it out later. All other things looks good so far.

harbaum, 2024-01-24:
I did not do a release yet, so nothing to test. I have so far only updated the tool chain.

dna2496, 2024-01-24:
ah thought you updated it already. ok i be patient

vanfanel, 2024-07-10:
Hi,

I would like to confirm that the only mice I have here (Logitech M150, 8BitDo N30) both exhibit the same strange behavior: making a continuous horizontal movement, the pointer goes back to previous positions, then advances again, etc.

This can be easily seen in Atari ST Hi-Res mode.

SanguineBrah, 2024-09-09:
This happens with all my mice & trackpads. The behaviour seems to be the same in every core, and it also seems to be the same whether I use the BL616 or RP2040 companion so I don't think CherryUSB is to blame.

It only seems to happen when the mouse is moved very fast across the screen, and it is the same in all directions. Perhaps it is something to do with the way large values are clamped in hid.c