MiSTle-Dev/MiSTeryNano #19: ACSI support

TzOk83, 2024-01-21:
I have tried 4 different hard disk images with the same result: 4 bombs.

Images I've tried:
http://www.mediafire.com/file/f1yq3z38741vwgx/SatanDisk2Gb_v1.rar
https://kerlinux.org/public/Atari_ST_Games_Petari_20181212_ACSI_1_6GB_5+4x400MB_ICD.zip
and 2 blank images from STEEM 4 (MEGAFILE 60, SH204).

I've renamed them from .img to .hd, all are raw binary images with some HDD driver installed.

harbaum, 2024-01-22:
Does AHDI or Hddriver or anything else loaded from floppy disk see a harddisk?

harbaum, 2024-01-22:
Ah ... do you use TOS 2.06? I see something like that with that version as well. Please try TOS 1.62 or TOS 1.04

harbaum, 2024-02-01:
The TOS 2.06 ACSI boot issue is now fixed and will be part of the next release

harbaum, 2024-02-01:
> Yes, I was using STE & TOS 2.06, with ST & TOS 1.4 it works, but I also tried PPhdrv 1.01hs and it works extremely slow; opening a folder in GEM takes well over 15 sec.
> 
> The RetroWiki image gives me 2 bombs at boot.

What is pphdrv 1.01hs ? And what is "the retrowiki image" ?

TzOk83, 2024-02-01:
pphdrv 1.01hs is Peter Putnik HDD Driver v1.01HS (2014), and RetroWiki Image is one of the images I gave link to in the first post (SatanDisk2Gb_v1.rar).

harbaum, 2024-02-01:
It's difficult to debug Peters stuff as it's closed source and not documented. Someone would need to disassemble e.g. the boot sector of that image and try to figure out what that does different than all other drivers.

closed 2024-02-02 by harbaum

harbaum, 2024-02-02:
The ppera driver relies on a hardware flaw in the DMA in the handling of the A1 signal. I traced the IO access of the ppera driver in Hatari, replayed this in my simulation of the ACSI chip, had the same broken behaviour. Now seeing the issue I knew where to look in hatari to see how they deal with this and found this comment:

https://github.com/hatari/hatari/blob/97eddba41d12661b0218850c160949e8510a6caa/src/hdc.c#L1139-L1141

harbaum, 2024-02-02:
Fixed in release 1.2.4 ... please give it a try

TzOk83, 2024-02-02:
Thank you for this release. The old PPDRV 0.98 (2009) works ok now, but the slightly newer version 1.01 (2014) still crashes (no disk access at all in GEM or sees a disk but with 0 bytes in size). Please take a look at the enclosed disk image.
[PP60.zip](https://github.com/harbaum/MiSTeryNano/files/14143878/PP60.zip)

On the bright side, I see a major improvement in USB mice/keyboard compatibility (yet, my gamepad still doesn't work). The mouse pointer is now much more responsive (less laggy), but fast movements make Atari lose the "track" of mouse pulses.

harbaum, 2024-02-02:
I'll have a look at that version. But the old version was already pretty broken ...

harbaum, 2024-02-02:
What gamepad are you trying to use?

TzOk83, 2024-02-02:
PiHut SNES: https://thepihut.com/products/raspberry-pi-compatible-usb-gamepad-controller-snes-style

harbaum, 2024-02-05:
> PiHut SNES: https://thepihut.com/products/raspberry-pi-compatible-usb-gamepad-controller-snes-style

I have a very similar cheap one. Will give it a try. You can of course always donate one and I'll see what I can do.

harbaum, 2024-02-05:
I think I found the issue the the later ppera driver. Since I didn't want to do a full release for this small change I just updated the existing release 1.2.4

So please re-download 1.2.4 and give it a try,

Unixware, 2024-05-07:
Just my 2p here: PPhdrv is bit buggy... I had issues with that even on my past MiST ...  ICD pro works fine with MiSTeryNano plus is way faster

commit 7db43bff (2024-02-02): Ignore A1 on second ACSI byte. Fixes #19