MiSTle-Dev/MiSTeryNano #34: How to flash docking station

tfanatic44, 2024-03-27:
Hello, I would like to know how to flash the docking station, whether when flashing the docking station, whether the docking station must be connected to the tang nano 20k module at the time of flashing, there is no information on this, or whether it is also possible to flash the docking station by directly connecting it to the computer  If so, how can the docking station be directly connected to the computer? Moreover, I have not found any information on how to connect the docking station with the tang nano 20k module

harbaum, 2024-03-27:
What dockingstation are you talking about?

tfanatic44, 2024-03-27:
M0S based on BL616

harbaum, 2024-04-24:
You just flash the M0S as you would without docking station

tfanatic44, 2024-04-24:
do you know how to cut a piece of the PCB with MIDI sockets, which was made by JLCPCB, you wrote that it is enough to break off a piece of the PCB with the USB-c socket to connect the M0S, I have this board but I am afraid to do it because I think it will break  all

harbaum, 2024-04-24:
I use a pair of side cutters and then smoothen the remains with a file.

I really think I should have omitted that part altogether ...

tfanatic44, 2024-04-24:
can you make a firmware for tang nano 20k that will use its internal USB C interface so as not to connect M0S to it, then you wouldn't have to break off a piece of the PCB with the USB c connector, do you plan to continue future using additional functions at all  M0S such as Wi-Fi 6 or Bluetooth, if it is not necessary M0S is unnecessary

harbaum, 2024-04-25:
You can yourself. The code in the repository is already prepared for this. But you need to deal with the fact that the hard setup is slightly different and that you need to find a way of dealing with FPGA updates etc.

Yes, the M0S is not necessary if I/we/you find a way around all those small issues that arrive from using the TN20k's own BL616.

tfanatic44, 2024-04-29:
I bought an electric mini saw and cut this piece of PCB, but unfortunately when I select any disk from the OSD menu, whether it's an A disk or an ACSI disk, the computer freezes and I can't do anything. Do you have any idea how to run a game?

harbaum, 2024-04-29:
Hmm, this connection is not the one being closest to the part where you cut. So I wouldn't expect that you damaged it since other more vital connections would then be damaged as well and as you say the OSD works.

Still, please use a multimeter to check the connections. 

Did you use a breadboard setup before? Did that work?

harbaum, 2024-04-29:
Can you select disk images from the menu? Then SD card IO generally works and so does IO14/IRQ. 

You select a disk and then the computer freezes? Did you try to select "Reset" from the OSD after you selected a disk image? The disk stays inserted if you select "reset" and the ST should boot from the newly selected floppy. Also what TOS are you using?

tfanatic44, 2024-04-29:
I am using the English version of tos, the version number you recommended on your website