MiSTle-Dev/MiSTeryNano #36: error when flashing the M0S dock station based on BL616/618, what to do

tfanatic44, 2024-04-05:
Unfortunately, when flashing BL616 on the M0S dock station, I receive an error. Am I doing something wrong? Can someone explain to me what's wrong?

[21:53:42.926] - serial type is general
[21:53:42.965] - Open COM3 Success
[21:53:43.979] - Close COM3 Success
[21:53:43.998] - EFUSE_CFG
[21:53:43.999] - BOOTHEADER_GROUP0_CFG
[21:53:44.034] - Create bootheader using C:\Users\Tomasz\Downloads\Atari St Misterynano\BouffaloLabDevCube-v1.8.9\chips\bl616\img_create_mcu/efuse_bootheader_cfg.ini
[21:53:44.036] - Updating data according to <C:\Users\Tomasz\Downloads\Atari St Misterynano\BouffaloLabDevCube-v1.8.9\chips\bl616\img_create_mcu/efuse_bootheader_cfg.ini[BOOTHEADER_GROUP0_CFG]>
[21:53:44.043] - Created file len:256
[21:53:44.050] - Create efuse using C:\Users\Tomasz\Downloads\Atari St Misterynano\BouffaloLabDevCube-v1.8.9\chips\bl616\img_create_mcu/efuse_bootheader_cfg.ini
[21:53:44.053] - Updating data according to <C:\Users\Tomasz\Downloads\Atari St Misterynano\BouffaloLabDevCube-v1.8.9\chips\bl616\img_create_mcu/efuse_bootheader_cfg.ini[EFUSE_CFG]>
[21:53:44.060] - Created file len:512
[21:53:44.069] - Image create path: C:\Users\Tomasz\Downloads\Atari St Misterynano\BouffaloLabDevCube-v1.8.9\chips\bl616\img_create_mcu
[21:53:44.070] - Config file: C:\Users\Tomasz\Downloads\Atari St Misterynano\BouffaloLabDevCube-v1.8.9\chips\bl616\img_create_mcu/img_create_cfg.ini
[21:53:44.089] - img already have bootheader
[21:53:44.092] - Image hash is b'94745365c4b866c44fce871c76e508928ad7c491758124c666ed60abf97c4051'
[21:53:44.093] - Header crc: b'7be7f464'
[21:53:44.094] - Write flash img
[21:53:44.098] - Encrypt efuse data
[21:53:44.108] - ========= eflash loader config =========
Traceback (most recent call last):
File "core\bflb_mcu_tool.py", line 2538, in program_img_thread
File "core\bflb_mcu_tool.py", line 2487, in program_bl616_img
File "libs\bflb_img_create.py", line 85, in compress_dir
File "libs\bflb_img_create.py", line 56, in check_pt_file
IndexError: list index out of range

harbaum, 2024-04-06:
I can't tell you as its the bouffalo tool that fails. Maybe the files you are trying to flash weren't downloaded properly and the tool failes to interpret them?

tfanatic44, 2024-04-06:
Is there any other program for Windows 10 to flash it correctly?

harbaum, 2024-04-06:
I don't know. But I don't think so.

tfanatic44, 2024-04-06:
Hello, I managed to flash , the error which involved downloading a file with errors from the project's website via a PC, so I downloaded this file to a mobile phone with Android and sent it to the PC using a USB cable. I immediately managed to flash the integrated circuit on the M0S board, thank you  you don't need help anymore

tfanatic44, 2024-04-08:
Hello, I solved this problem by downloading the M0S firmware file from the GitHub website from an Android phone and then transmitting the file via a USB cable to the PC.  , but a friend from Buffalo gave me another solution
Hi
    The path of the tool should not have spaces

    C:\Users\Tomasz\Downloads\Atari St Misterynano\BouffaloLabDevCube-v1.8.9

    change to  C:\Users\Tomasz\Downloads\AtariStMisterynano\BouffaloLabDevCube-v1.8.9

￼

jxtan

jxtan@bouffalolab.com

harbaum, 2024-04-08:
Upated the INSTALLATION_WINDOWS.md accordingly. Thanks for reporting.

closed 2024-04-08 by harbaum