MiSTle-Dev/MiSTeryNano #37: 'blt_clks' is not declared

Chandler-Kluser, 2024-04-07:
# Issue

Can't build bitstream using `Gowin_V1.9.8.11_Education_linux` through command in `/src` folder:

```
$ /home/chandler/Gowin_V1.9.8.11_Education_linux/IDE/bin/gw_sh ./build.tcl
```

## Expected result

Gowin Shell fully synthesize, place and route the bitstream `.fs` file

## System specs

```
$ neofetch       
                   -`                    chandler@archlinux 
                  .o+`                   ------------------ 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: B450 I AORUS PRO WIFI 
               `+oooooo:                 Kernel: 6.8.2-arch2-1 
               -+oooooo+:                Uptime: 1 day, 15 hours 
             `/:-:++oooo+:               Packages: 1406 (pacman) 
            `/++++/+++++++:              Shell: zsh 5.9 
           `/++++++++++++++:             Resolution: 3840x2160 
          `/+++ooooooooooooo/`           DE: GNOME 46.0 
         ./ooosssso++osssssso+`          WM: Mutter 
        .oossssso-````/ossssss+`         WM Theme: Adwaita 
       -osssssso.      :ssssssso.        Theme: HighContrastInverse [GTK2/3] 
      :osssssss/        osssso+++.       Icons: Adwaita [GTK2/3] 
     /ossssssss/        +ssssooo/-       Terminal: kgx 
   `/ossssso+/:-        -:/+osssso+-     CPU: AMD Ryzen 7 3700X (16) @ 3.600GHz 
  `+sso+:-`                 `.-/+oso:    GPU: AMD ATI Radeon RX 6600/6600 XT/6600M 
 `++:.                           `-/+/   Memory: 4833MiB / 15912MiB 
 .`                                 `/
```

## Error details

During Synthesis, I have got this error message:

```
$ /home/chandler/Gowin_V1.9.8.11_Education_linux/IDE/bin/gw_sh ./build.tcl | grep ERROR   
ERROR (EX3872) : 'blt_clks' is not declared("/home/chandler/Documentos/MiSTeryNano/src/atarist/atarist.v":666)

    while executing
"run all"
    (file "./build.tcl" line 101)
```

## Full log

This is the TCL Shell output during build process:

```
*** GOWIN Tcl Command Line Console  *** 
current device: GW2AR-18C  GW2AR-LV18QN88C8/I7
add new file: "misterynano.sv"
add new file: "atarist/acia.v"
add new file: "atarist/acsi.v"
add new file: "atarist/atarist.v"
add new file: "atarist/dma.v"
add new file: "atarist/io_fifo.v"
add new file: "atarist/mfp.v"
add new file: "atarist/mfp_hbit16.v"
add new file: "atarist/mfp_srff16.v"
add new file: "atarist/mfp_timer.v"
add new file: "atarist/stBlitter.sv"
add new file: "atarist/ste_joypad.v"
add new file: "atarist/cubase2_dongle.v"
[... 202 lines cut ...]
ERROR (EX3872) : 'blt_clks' is not declared("/home/chandler/Documentos/MiSTeryNano/src/atarist/atarist.v":666)
WARN  (EX3073) : Port 'floppy_ready' remains unconnected for this instance("/home/chandler/Documentos/MiSTeryNano/src/atarist/atarist.v":899)
Module 'misterynano' remains a black box due to errors in its contents("/home/chandler/Documentos/MiSTeryNano/src/misterynano.sv":8)
Compiling module 'video2hdmi'("/home/chandler/Documentos/MiSTeryNano/src/tang/nano20k/video2hdmi.v":6)
Compiling module 'pll_160m'("/home/chandler/Documentos/MiSTeryNano/src/tang/nano20k/gowin_rpll/pll_160m.v":10)
Compiling module 'Gowin_CLKDIV'("/home/chandler/Documentos/MiSTeryNano/src/tang/nano20k/gowin_clkdiv/gowin_clkdiv.v":9)
Compiling module 'hdmi(AUDIO_RATE=48000,VENDOR_NAME=64'b0100110101101001010100110101010001101100011001010000000000000000,PRODUCT_DESCRIPTION=128'b01000001011101000110000101110010011010010010000001010011010101000000000000000000000000000000000000000000000000000000000000000000)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/hdmi.sv":8)
Compiling module 'packet_picker(VIDEO_RATE=32000000.0,IT_CONTENT=1'b1,AUDIO_BIT_WIDTH=16,AUDIO_RATE=48000,VENDOR_NAME=64'b0100110101101001010100110101010001101100011001010000000000000000,PRODUCT_DESCRIPTION=128'b01000001011101000110000101110010011010010010000001010011010101000000000000000000000000000000000000000000000000000000000000000000,SOURCE_DEVICE_INFORMATION=8'b00000000)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/packet_picker.sv":4)
Extracting RAM for identifier 'audio_sample_word_buffer'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/packet_picker.sv":81)
Compiling module 'audio_clock_regeneration_packet(VIDEO_RATE=32000000.0,AUDIO_RATE=48000)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/audio_clock_regeneration_packet.sv":5)
Compiling module 'audio_sample_packet(SAMPLING_FREQUENCY=4'b0010,WORD_LENGTH=4'b0010)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/audio_sample_packet.sv":8)
WARN  (EX3780) : Using initial value of 'CHANNEL_LEFT' since it is never assigned("/home/chandler/Documentos/MiSTeryNano/src/hdmi/audio_sample_packet.sv":63)
WARN  (EX3780) : Using initial value of 'CHANNEL_RIGHT' since it is never assigned("/home/chandler/Documentos/MiSTeryNano/src/hdmi/audio_sample_packet.sv":64)
Compiling module 'auxiliary_video_information_info_frame(IT_CONTENT=1'b1)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/auxiliary_video_information_info_frame.sv":5)
Compiling module 'source_product_description_info_frame(VENDOR_NAME=64'b0100110101101001010100110101010001101100011001010000000000000000,PRODUCT_DESCRIPTION=128'b01000001011101000110000101110010011010010010000001010011010101000000000000000000000000000000000000000000000000000000000000000000,SOURCE_DEVICE_INFORMATION=8'b00000000)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/source_product_description_info_frame.sv":5)
Compiling module 'audio_info_frame'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/audio_info_frame.sv":5)
WARN  (EX1998) : Net 'headers[255][23]' does not have a driver("/home/chandler/Documentos/MiSTeryNano/src/hdmi/packet_picker.sv":30)
WARN  (EX1998) : Net 'subs[255][3][55]' does not have a driver("/home/chandler/Documentos/MiSTeryNano/src/hdmi/packet_picker.sv":31)
Compiling module 'packet_assembler'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/packet_assembler.sv":4)
Compiling module 'tmds_channel'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/tmds_channel.sv":4)
Compiling module 'tmds_channel(CN=1)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/tmds_channel.sv":4)
Compiling module 'tmds_channel(CN=2)'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/tmds_channel.sv":4)
Compiling module 'serializer'("/home/chandler/Documentos/MiSTeryNano/src/hdmi/serializer.sv":3)
GowinSynthesis finish
```

harbaum, 2024-04-08:
blt_clks is defined in stBlitter.sv and is included. Please try a more recent version of the GoWin suite.

Chandler-Kluser, 2024-04-08:
actually I have not found any directive in `atarist.v` including stBlitter

Switched to v1.9.9 and it worked with some modifications. For code sanity, if you let me make some suggestions to avoid issues like that in future:

1. in `build.tcl` you can add `stBlitter.sv` before `atarist.v`, since it has been defined there, or
2. define the struct and params associated with blitter clock in `atarist.v` and include in `stBlitter.sv` through a directive

Both approaches are synthesizeable using old IDEs. In v1.9.9 I fully built using the first method.

If you want, I can make a PR with this little modification.

closed 2024-04-08 by Chandler-Kluser

harbaum, 2024-04-10:
Implemented variant 1. Thanks.