MiSTle-Dev/MiSTeryNano #38: MisteryShield 20K Bom error (R9)

tophe38, 2024-04-23:
Hi,

I found an error in this BOM
https://github.com/harbaum/MiSTeryNano/blob/main/board/misteryshield20k/jlcpcb/bom.csv

R9 shoud be a 270 ohm resistor, but C17530 is a 2.7K Ohm one.

C17590 should be the correct one.

harbaum, 2024-04-23:
Interesting. Thanks for reporting. I'll update the data asap ...

closed 2024-04-24 by harbaum

harbaum, 2024-04-24:
This is now fixed but it should not affect operation. This is just a pull-up on the MIDI RX line going to the FPGA and even 2k7 should pull fast enough for MIDI data to be received correctly. If in doubt you can connect MIDI In with MIDI OUT and test the loopback e.g. with the Rufus terminal program.

tophe38, 2024-04-24:
Thanks.
I'm using a 330 ohm resistor, and will test it.

commit f73f1d8a (2024-04-24): Fix #38