MiSTle-Dev/MiSTeryNano #40: failing to install the nano (20K variant)

Unixware, 2024-04-24:
This is what I m getting when I try:

```
[bin]: ./openFPGALoader -f atarist.fs 
empty
write to flash
No cable or board specified: using direct ft2232 interface
unable to open ftdi device: -4 (usb_open() failed)
JTAG init failed with: unable to open ftdi device
```

is there a 'different' variant of 20K that is supported? There is a silkscreened 'v3921' on my board (version/rev probably..) thanks

closed 2024-04-24 by Unixware

harbaum, 2024-04-24:
You don't necessary need root permissions. A matching udev rule may be sufficient. See e.g. https://wiki.sipeed.com/hardware/en/tang/Tang-Nano-Doc/flash-in-linux.html

harbaum, 2024-04-24:
I've added a link to the Sipeed page to the installation instructions