MiSTle-Dev/MiSTeryNano #42: misterynano_fw commit ID c378344 broken related to proper Keyboard detection

vossstef, 2024-05-07:
facing problems in proper USB Keyboard detection after re-compile the sources for C64 and VIC20 core use.

- Keyboard very often not properly detected. Had to unplug and re-connect the HID to overcome.
- In usb_host.c the VIC_20 entries are lost...

I switched lately to latest Bouffalo SDK+ CherryUSB 1.2 and initially thought that problems arise from that but figured out that it is the FW. 

As back-up i included my needed menu entries and spi loader fixes in a branch based on cb65679. New SDK + FW without all that BLE and HID enhancements are working rock solid stable.

harbaum, 2024-05-08:
Oops ...

My keyboards are still working. Can you tell which exact change causes your problems? Do you see anything suspicious in the serial debug output?

vossstef, 2024-05-11:
I send the logs via PM. Checked that there is no Problem with a normal USB Keyboard and incompatibility is related to my 
old Logitch K600 Keyboard combo only. 
https://www.logitech.com/en-us/products/keyboards/k600-smart-tv-d-pad.920-008822.html

Maybe let's close the topic and i simply buy more standard type of wireless KB...

closed 2024-05-11 by vossstef

harbaum, 2024-06-01:
I am considering undoing the Bluetooth part. Since there are very few pure ble hid devices and it broke things.

Also the SDK support is pretty bad and e.g classic Bluetooth is basically unusable.