MiSTle-Dev/MiSTeryNano #44: "Sea of Colour" demo shows a black screen

vanfanel, 2024-05-07:
Hi there,

Just tried this core on the Tang Nano 20k for the first time, and I noticed that the demo "Sea of Colour" just shows a black screen.

Here is the demo in question to download:
https://www.pouet.net/prod.php?which=65921

Maybe an HDMI scaler problem? I tried many other demos and they all seem fine.

vanfanel, 2024-05-08:
The demo works perfectly well on MiSTer using the same core, so my guess it's something with the Tang Nano 20k port.

harbaum, 2024-05-08:
What hardware are you using? Are you using a breadboard setup?

ojaksch, 2024-05-08:
I can confirm that the demo isn't working, here tested with TOS 1.06/de and using your misteryshield20k/M0S board.

vanfanel, 2024-05-09:
> What hardware are you using? Are you using a breadboard setup?

I'm using this "docker", which has integrated Host usb M0S, usb-c and DB9 with protection:
[image]

It was made in limited numbers recently with your permission.

But I don't think it's the cause at all :)

vanfanel, 2024-05-09:
Hmm.. Maybe I should update the M0S Dock firmware? The C64 core seems to require that.
If that's the case, is it possible to update the M0S Dock firmware on GNU/Linux?

harbaum, 2024-05-09:
> But I don't think it's the cause at all :)

No, it's sure not. But you might want to ask those for support first, that sell you the hardware.

vanfanel, 2024-05-09:
> > But I don't think it's the cause at all :)
> 
> No, it's sure not. But you might want to ask those for support first, that sell you the hardware.

Those who sold me the hardware don't develop the core, and this seems like a core issue. Does the demo work for you?

harbaum, 2024-05-09:
The core is based on many people's work. But none of them is making money with it. The vendors are the ones making money and it would IMHO be fair if they also help fixing problems.

I haven't tried to reproduce the problem, yet. I am currently working on something different.

harbaum, 2024-06-06:
I got the files from pouet, used hatari to copy them onto a hdd image using hdddriver 12.03. The resulting installation gives some garbled graphics on hatari as well as on misterynano. Are there ready-to-run images for this demo?

vanfanel, 2024-06-06:
@harbaum Yes, I have uploaded an small HDD image ready to use, that contains several demos, with Sea of Colours among them:

[ST-Demos.pdf](https://github.com/user-attachments/files/15614470/ST-Demos.pdf)

It's a .7z file, so rename it's extension to .7z to be able to decompress it.
(I had to rename it to .pdf because github doesn't allow uploading .7z for some absurd reason).

vanfanel, 2024-06-06:
Also, the file is here:

https://drive.google.com/drive/folders/1hEaq-g87sdXDVN3_eGf0hUojHI-UBh7c

Just rename .vhd to .hd to use it with the core.

ojaksch, 2024-06-06:
Hehe, these are my demo images, originally made for the MiSTer - but I knew that they might be needful :1st_place_medal:

harbaum, 2024-06-06:
The sound indeed runs. But I am not sure it's just the black screen. IMHO the demo should continue after a few seconds and the birds noise should stop which it doesn't. So it seems like the demo engine itself fails.

The readme says that one needs a HDD that gives at least 500kB/s. I am not sure the MiSTeryNano achieves that. What happens if the disk is too slow?

ojaksch, 2024-06-06:
HDD speed might be a culprit. Just tested a 5MB image/HDDriver 11 (AHCI speedup enabled) with [AHPT9](https://atari.8bitchip.info/ahpt.html) and results are low:

[image]

vanfanel, 2024-06-08:
What's the limiting factor here? The disk controller implementation?
Modern MicroSDs are way faster than 5 MB/s...

luciust, 2025-03-29:
I can confirm the same issue - music is fine, black screen. On the other hand the https://www.youtube.com/watch?v=mlmrPPBrr6Q works fine and it also needs fast HDD.