MiSTle-Dev/MiSTeryNano #46: F12 not working?

rootuser, 2024-05-14:
Hello,

thank you for this great project! The Atari ST was a big part of my youth and I was delighted when I read about your project in C't.

Alas, I have difficulties getting the project to fully work.

As far as I can tell I have set up everything correctly according to the installation manual for Windows users. To be on the safe side, I'll list everything I've done:

1. I used the Gowin IDE to flash firmware and two TOS versions (1.04 and 1.62, both german) to the Tang nano. I used the correct settings (different Operations for firmware and TOSes, different start addresses). I flashed the three files one by one.
I also used the 'press S2 button' method for connecting the tang. What I noticed though (and that is missing in the installation manual): When one file was flashed, the Tang seemed to restart and the two 'disc activity' LEDs lit up. So I always had to unplug the Tang and re-connect it with S2 button pressed.
Flashing always finished without errors
2. I flashed the MOS-Dock, following all of yout instructions, having no trouble or errors shown.
3. After that, I followed the trouble shooting guide. Step 1 and 2 obviously worked, I got a nice Atari desktop. Neat!
4. At this step, I connected everything to the MiSTeryNano Shield: original USB power supply removed and replaced by MOS dock, Tang nano in place.

From here, I am not sure if it is working or not. In step 3 of the troubleshooting guide you say: 'the [...] LED {will change to} to blue and later to green or red.' On my setup, the LED will always just be red and does not change at all. On the other hand you then say that 'if the [...] LED [...] lights up green, red or blue', everything is fine.
Everything _seems_ fine at first look; when I connect my Keyboard/mousepad combi  (Logitech k400r), the touchpad works nicely. 

But when testing further I find: inserting a SD card also does nothing to the LED Color, and pressing F12 does not show the onscreen menu. Concerning SD card I have already followed your instruction in [https://github.com/harbaum/MiSTeryNano/issues/20](url)

At this stage, I cannot find any information on what I might have done wrong or could try to fix the issue.
Would you kindly help?

Thank you!

harbaum, 2024-05-14:
The troubleshooting guide does _not_ say that everything is fine when the LED lights up red. It says that the M0S is properly flashed and _most_ connections are fine. And that's true as it's the M0S that controls the LED. And if it runs into trouble when trying to read the SD card then it switches the LED to red.

I am actually not sure if the F12 should still work in that case. But that would not help you much as you need to get the SD card working, anyway.

The red LED says that the SD card could no be accessed. That's the problem you need to solve. Trying to open the menu using F12 will not solve the SD card problem. Please try a different SD card or try to reformat the one you have.

rootuser, 2024-05-15:
Hello,

thank you very much for your reply. 

I really tried everything. 
Firstly, I flashed everything again (using each Windows AND Linux). No change. 
Then I began experimenting with different SD cards (2 to 128GB), formatted in Windows and Linux. No change. 
When I did not achieve anything with it, I suspected faulty hardware. So flashed a new TangNano. No change. 
I flashed and used a new M0S dock. Nothing. 
Again I tried all SD cards. Still nothing.

Then I changed to a different MisteryShield and finally it worked AT ONCE. Phew.
Turns out there is a broken signal line on the circuit board (MISO), which prevented the M0S dock from communicating correctly with the Tang nano.

Now everything works perfectly and the quality of the realisation simply blows me away.
Thank you very much for this unbelievable project!

By the way: now that it works, every SD card I have used so far has been recognised. At the moment I am using a 64GB card with exFat format, for example.

Regards, Tom

closed 2024-05-15 by rootuser

harbaum, 2024-05-15:
Excellent! Have fun with your new Atari ST.