MiSTle-Dev/MiSTeryNano #49: feature idea BL616 Firmware: Analog Paddle Emulation via USB Gamepad Stick(s)

vossstef, 2024-05-28:
feature idea:

So far digital USB Joystick is supported. 

Replacement of existing DualShock 2 **analog Stick mode** used presently for VIC20 C64 paddle emulation with 
e.g. [USB Logitech Gamepad](https://www.logitechg.com/en-us/products/gamepads/f310-gamepad.940-000110.html) or of course other cheaper brand.

Not sure is such mode supported at all...

harbaum, 2024-06-01:
This should easily be possible. The hid code already reads analogue values if present and currently maps them into digital ones.

I'd suggest to add a status info in the FPGAs hid side which states which joystick data is preferred. Also future cores may want more axes and buttons.

vossstef, 2024-09-05:
implemented in FPGA Companion firmware

closed 2024-09-05 by vossstef