MiSTle-Dev/MiSTeryNano #51: Firmware roadmap

harbaum, 2024-06-01:
There are a few things I'd like to add/change in the firmware:

- remove/disable the not really useful Bluetooth code. This is due to the fact that the bouffalo SDK does not properly support classic Bluetooth and very few Bluetooth hid devices support pure BLE

- move core specific code from menu.c into separate files

- remove core specific core entirely. Instead have all core specifics (menu, variables, storage types...) in a separate config file which is either read from the core or from sd card. This will eventually allow to add new cores without recompiling the firmware.

- add wifi/tcp/telnet support. This could be mapped onto a serial port on core side and would allow cores to dial into BBS's which today often use the same approach to expose a dial up like connection via the internet

harbaum, 2024-06-12:
The more functionality is added to the BL616/M0S the more the SDK becomes of a problem. Thus rather high on the list now is:

  - make the firmware device independent and make it e.g. run on a raspberry pi pico, esp32 or the like

harbaum, 2024-06-20:
The FPGA Companion project will replace the MiSTeryNano firmware:
  https://github.com/harbaum/FPGA-Companion

It's now seperate as it's also used by several other cores like C64, VIC20 and Minimig/NanoMig

The FPGA Companion can also be built for the Raspberry Pi Pico/RP2040 and the Pico can replace the M0S in a MiSTeryNano setup.

mega-hz, 2024-09-12:
Will be there a difference in running the ST?
Or just a new SourceCode Workaround?