MiSTle-Dev/MiSTeryNano #53: SD Card

MATRIX7878, 2024-06-08:
Hello,

     I wanted to confirm a few things.

1. This project uses the SD card in SD 4-bit mode
2. I cannot immediately find the SD card calls.  I see the three files in the misc folder, but I cannot find the sd_reader file.  Where do you call the functions to write and read to the SD card?

Thank you

harbaum, 2024-06-12:
See https://github.com/harbaum/MiSTeryNano/blob/main/src/misc/sd_rw.v

MATRIX7878, 2024-06-12:
Ah, so it is in 4-bit SD mode.  I see that that file is for reading/writing.  I was asking for an example where data was being transferred.  I guess I was foolishly thinking of seeing a block/multiblock read and write example in the code somewhere.  Thank you though.

closed 2024-06-18 by harbaum