MiSTle-Dev/MiSTeryNano #58: How can i use Hard Disk Images?

jbrunk1966, 2024-06-26:
The Atari-ST Emulation is running well and i can select several Disk Images via the OSD but how can i create a Hard Disk Image pls? AFAIK in TOS 1.x the max Partition Size is 256 MB and in 2.x 512 MB, is this sill here also the Limit?

harbaum, 2024-06-26:
If you are not familiar with the usage of harddisks on the ST I'd suggest that you start by using an emulator like Steem or Hatari and create a harddisk setup there following a tutorial. The resulting harddisk image can then be copied to the sd card to be used with the MiSTeryNano as well.

And yes, since it's a hardware remake all limitations of the real machine also apply to this. The size of harddisk images varies between the TOS versions and the harddisk driver software.

harbaum, 2024-06-26:
Does the image actually work inside Hatari? Can it do there what you want it to do on the MiSTeryNano? Is it formatted, partitioned, did you install a harddisk driver? Please make sure it works in Hatari.

jbrunk1966, 2024-06-26:
> For STE mode a STE capable TOS needs to be flashed to a different flash location:
okay that could explain the Screen Issue im STE mode, will flash a STE TOS then :crossed_fingers:

harbaum, 2024-06-26:
The STE TOS is to be flashed to a different flash location. See the instructions.

jbrunk1966, 2024-06-26:
> Do you have installed a harddisk software onto the image? That is a must to use the image as a boot device in an emulator or FPGA. Also keep in mind that the tool `atari-hd-image` only creates a DOS partition scheme image, which isn't compatible with TOS (but can be accessed with Uwe Seimet's HDDRIVER).

it only wotks in EmuTOS not with real TOS ROM Images. So pls how can i create a TOS compatible Harddisk Image and which free Tools / Driver do i need to run them with MiSTeryNano? Thanks!

jbrunk1966, 2024-06-26:
found a prepared TOS compatible 512 MB [Disk Image](https://misterfpga.org/viewtopic.php?t=183) which runs well in Hatari and TOS 2.06 and seems to include already the needed driver. Will copy this over to a SD card and will test it in the FPGA asap :crossed_fingers:

jbrunk1966, 2024-06-26:
> found a prepared TOS compatible 512 MB [Disk Image](https://misterfpga.org/viewtopic.php?t=183) which runs well in Hatari and TOS 2.06 and seems to include already the needed driver. Will copy this over to a SD card and will test it in the FPGA asap 🤞

this works like a charme also in the FPGA, except even 2 drive got regonzied during boot just 1 is availiable in TOS :shrug:

ojaksch, 2024-06-26:
The problem may be how to add stuff from PC to that image. Personally I'm using the mentioned HDDRIVER which is capable of using TOS, DOS and mixed partition schemes (and more types, too). The "mixed" one can also be used to access the image from a PC by using the `mtools`.

HDDRIVER is a commercial but still developed tool, imho the best one and a nearly full usable demo version is available.
I can provide you an image in either size, ready2run with an installed demo version of HDDRIVER. Even if it's just to play around.

jbrunk1966, 2024-06-26:
> The problem may be how to add stuff from PC to that image. Personally I'm using the mentioned HDDRIVER which is capable of using TOS, DOS and mixed partition schemes (and more types, too). The [...]

Thanks a lot but's it's not necessary. I fiddled around with the Atari Hardisk Drive Install Disk and "fiound" the 2nd Diive in TOS after "Scan for Devices" in the GEM Menu ... anyway one drive is still enough to play with it. So tanks a lot for your help and your work and effort into this great Project!