MiSTle-Dev/MiSTeryNano #59: Cant find file for µC

mega-hz, 2024-07-23:
Hi,

i am trying to programm the µC for use as USB-HID but get this error:

[21:19:16.468] - Error: FW:Z:\MiSTeryNano-main\firmware\misterynano_fw\./misterynano_fw_bl616.bin image file is not existed

How to create this file or wher can i find it?

harbaum, 2024-07-23:
From the same place where you got the ini file:

https://github.com/harbaum/MiSTeryNano/releases

mega-hz, 2024-07-23:
ok, thank you.
But it was not in the whole zip...

So now it should work wit mouse, keyboard and usb-hub?

And another question: is LPT and SER already on Tang Nano 20K Pins available?

mega-hz, 2024-07-24:
So i reprogrammed the Onboard BL616 with that file, connected the TangNano to the Input of an USB-C Hub but neither Mouse nor Keyboard are working.
And when i connect Pin73 to GND it doesn't switch to Monochrom.

mega-hz, 2024-07-25:
Any Idea how to get this run with keyboard on builtin BL616?

harbaum, 2024-09-04:
I have dropped support for the built-in bl616. On some versions of the tang nano 20k it's connection is broken (a unnecessary capacitor limits the frequency too much), the firmware is now encrypted etc etc ...

closed 2024-09-04 by harbaum