MiSTle-Dev/MiSTeryNano #62: Resoltion on Monitor

mega-hz, 2024-09-13:
Hi there,

i wonder is it possible to change the Resolution to 1280x1024 instead of 1920x1080 ?
THAT would be great because my NEC Monitor doesn't allow more.

Is there a place in the sourcecode to change that?

harbaum, 2024-09-13:
> i wonder is it possible to change the Resolution to 1280x1024 instead of 1920x1080 ?

What makes you think it"s doing 1920x1080?

The MiSTeryNano is doing a scan doubled standard PAL or NTSC video timing which is e.g. 576p in case of PAL. This is nowhere near 1080p or 1024p.

mega-hz, 2024-09-13:
Well, my nec monitor accepst max. 1280x1024 and shows no picture when connected tot he nanoMister

harbaum, 2024-09-13:
Then please don't just claim it's doing 1080p. Bug reports based on guessed facts often lead into the wrong direction and makes people search for the wrong thing.

Looks like your screen doesn't support a scan doubled PAL signal. You might try a US TOA rom to see if it copes with a NTSC timing.

But ultimately there's not much you can do if your screen doesn't support that mode.

The screen resolution you are mentioning is typical for an older computer screen. Are you sure you are using a HDMI screen? Is is it a DVI screen using some  HDMI/DVI adapter cable?

mega-hz, 2024-09-13:
Yes it is HDMI 2 DVI
But none of my old monitors will accept that signal.
Only my „modern“ Monitors with can full hd or 4K.
 
So is there a solution to change the resolution in source code?
I really want to see the picture on my NEC Multisync 1970GX..
 
Regards,
Wolfram.

 

Looks like your screen doesn't support a scan doubled PAL signal. You might try a US TOS rom to see if it copes with a NTSC timing.

harbaum, 2024-09-14:
HDMI is not DVI. With that cable you are sending HDMI over DVI ... but it's still HDMI. Many old screens don't cope with that.

Some recent update in the MisteryNano's HDMI.sv source re-enabled the option to build the core with DVI rather than HDMI. You can give that a try. But if course you then don't have audio.

There are several other options. One is to use a real HDMI to DVI converter/rescaler. Another is to use a HDMI to VGA converter. Both usually also export analogue audio.

mega-hz, 2024-09-14:
ok. in wich part of the source code can i change it from HDMI to DVI?
And is there another possibility to get the sound?

harbaum, 2024-09-14:
You can change this in HDMI.sv as I wrote in my previous post.

mega-hz, 2024-09-14:
hm, where exactly can i find that to switch to dvi?
Line Nr.?

harbaum, 2024-09-14:
I confused that with the NanoMig which recently had this option added. The MiSTeryNano would need to get the same change.

But I don't see a need nor do I have a DVI screen to test. So I'll leave it for someone else to (re)implement DVI support.

closed 2024-09-14 by harbaum

mega-hz, 2024-09-14:
OK.
No problem. I was just wondering because a PI3 works over HDMI2DVI Adapter.
Never mind.
 
But i have a much greater Problem:
Mouse works fine, but Keyboard has real issui!
A cable-Keyboard only accepts ALT-Arrow-up and Strg-Alt-del. No more.
A Wireless Keyboard even doesnt work at all.
Both work w/o problems on a pc.
 
Any Idea?