MiSTle-Dev/MiSTeryNano #63: Does harddrive images work?

pawelka, 2024-10-01:
First of all, great project!!!

I tried to test harddisk image. In OSD I was able to select hd.img and I saw in longs (on companion device) that it's reading something (many "SDC: sdc_read(0x3fc99404,5403,1)"), after that nothing happen, no diskdrive. I also try to restart, but only same log during loading and nothing else.

Is harddisk image is implemented on FPGA? Or it's only beginning of implementation in MCU (OSD)?

I tried 1.04 and 2.06 TOSes and I tested image with Hatari emulator before trying it on FPGA.

ojaksch, 2024-10-01:
It's working fine since months. Not the fastest in the world, but working.
I'd check the SD card, wiring etc. I had a SD expansion cable while experimenting with the MiSTeryNano, while this is working in principle it turned out to be unreliable and failing sometimes. Changed it to known good and working SD card and ACSI was working fine again.

harbaum, 2024-10-01:
The logs should tell you which device it's reading from and whether it's the floppy or the HDD. In case of the HDD you need a harddisk driver installed in you image. If that's present and being loaded you should immediately see its boot messages on the Atari ST screen.

pawelka, 2024-10-11:
First image which I tested doesn't have driver in it, but it works with emutos because it has driver in tos. So I tested with standard image 1.04 on hatari and image without driver doesn't work (according harbaum comment this is expected behaviour). I have another HDD image 1.5GB with multiple HDD drives inside, but fpga is rebooted before image is loaded and it takes long time before reboot :-/  Maybe there is some watchdog causing that. This big image work without any issue on HAtari and TOS 1.04

Additionally emutos doesn't work with FPGA, I got panic error with wrong address :-/

pawelka, 2024-10-24:
I've tested additionally with BL616 and 1.5GB image works, a little bit slow during image selection, but after that works as expected. It looks like it's ESP32-S3 issue only.

pawelka, 2024-10-26:
ESP32-S3
I rebuilt the image with the sdc_debugf removal for each sector read, and the image loaded successfully, though its behavior is still a bit unusual. The image is 1.5 GB, so when selecting it in the OSD, the OSD blocks for a significant amount of time after displaying “Link table ok.” I saved the settings and restarted the ESP32 and FPGA, but it stalled in the COM FPGA thread, likely due to the image loading in the background. TOS loads, but the image isn’t visible. After reviewing the logs and seeing “Link table ok,” I tried restarting the FPGA alone, and the image then mounted successfully in TOS. It appears to be a race condition between the FPGA and MCU.