MiSTle-Dev/MiSTeryNano #70: Programming multiple files on the Nano 20K

SparkyNZ, 2024-11-09:
I've been on for several hours this afternoon trying to load MiSTeryNano on my 20K.

I can only load 1 file at a time to my 20K - it may be worth mentioning this in the programming/loading instructions. I've been battling with "Cable not found" and "Device not found" etc and all because I had multiple checkboxes ticked when trying to program :-)

harbaum, 2024-11-09:
Thanks for the report. I have never noticed since I basically always use openfpgaloader.

Please feel free to open a pr.

SparkyNZ, 2024-11-14:
I'm a bit limited on time but I'm happy to do raise a PR. My biggest obstacle at the moment is figuring how to create a new branch on Github so I can raise a PR.