MiSTle-Dev/MiSTeryNano #72: Max visible hard disk image size is 2579MB

TzOk83, 2024-12-14:
I have an ACSI hard disk image with Peter Putnik's games. It is ~6.5GB in size and has 14 partitions. Under MiSTeryNano I can see only the first 5 partitions and the reported disk size is 2.5GB (2579MB).

pawelka, 2024-12-15:
Which TOS you used? Improved or regular? I used Peter images with improved TOS and it works for me. SD initialization is quite slow and after loading requires soft restart.

harbaum, 2024-12-16:
The SD card is using a FAT file system which only supports files up to 2 GB. I haven't checked what exfat supports.

Does your PC actually see the full size of the file when stored on SD card?

harbaum, 2024-12-16:
Another aspect of the ppera drivers and images is that they are closed source and have to be paid for. Thus I don't have access to them and especially cannot debug problems with them easily.

TzOk83, 2024-12-19:
The image contains multiple 512MB partitions, but only the first 5 are visible, and PP HD Driver sees only 2579MB space on a hard disk. The problem is that the ACSI disk image size seems to have been truncated to 2579MB.
ExFAT has no reasonable size limitations (128PiB).

TzOk83, 2025-04-13:
I've managed to replicate this problem using a blank disk image. A 4 GiB image works fine, but this one, which is precisely 7,000,000,000 bytes long, does not.

Below is an empty 7GB image, which has 13 partitions. STeEM sees it correctly:
[image]
but MiSTeryNano can see only the first 5 partitions and a disk size of 2579MB:
[image]

[7GB.zip](https://github.com/user-attachments/files/19727144/7GB.zip)

TzOk83, 2025-05-10:
The original image size is 6675MB, MiSTery sees it as 2579MB, so somewhere we have a 4GB rollover, as 6675-2579=4096.

harbaum, 2025-08-12:
Fixed in repository. Will be part of the next binary releases.

[image]

closed 2025-08-12 by harbaum